EarToEarOak/RTLSDR-Scanner #12: App crash after changing GPS type

keerah, 2015-08-22:
Each time I change GPS type to any other than default the application crashes immediately or after applying changes. Running as admin doesn't help.

EarToEarOak, 2015-08-27:
Thanks
Could you post the error message, I've been unable to replicate the problem?

keerah, 2015-08-27:
I only have a message that Python's been stopped, then the program exits
[image]
Sorry it's russian localization, it says python.exe has been stopped, gathering additional info

EarToEarOak, 2015-08-28:
Thanks, could you try one of the [stand-alone](https://github.com/EarToEarOak/RTLSDR-Scanner/releases/latest) executables and see if you get the same error?

keerah, 2015-08-28:
Sorry, mates, I couldn't even run it. At first it asked for librtlsdr to be added to PATH. Once I've done it, it asked for the same. Then I've moved the executable into librtlsdr folder, run it from there, and got this 
[image]

keerah, 2015-08-28:
Repeated with 32bit version. Worked fine! Gonna test it further.

closed 2015-10-14 by EarToEarOak