EarToEarOak/RTLSDR-Scanner #13: rtlsdr_scan.py GUI is unresponsive while scanning.

oddstr13, 2015-09-14:
The RTLSDR Scanner GUI is unresponsive while scanning.

 * Windows 7
 * Python 2.7
 * RTLSDR-Scanner-1.0.21972.32419.zip

pip freeze
------------
```python
arrow==0.6.0
BeautifulSoup==3.0.8.1
beautifulsoup4==4.2.1
cheetah==2.4.4
Cython==0.18
Django==1.6.5
django-appconf==0.6
django-compressor==1.3
django-fiber==0.13
django-mptt==0.6.0
djangorestframework==2.3.8
easy-thumbnails==1.4
esky==0.9.7
fs==0.5.2
Genshi==0.7
lupa==0.20
lxml==3.3.1
markdown==2.4
matplotlib==1.3.1
numpy==1.6.1
paramiko==1.10.1
PIL==1.1.7
Pillow==2.2.1
psutil==0.7.0
py2exe==0.6.9
pycairo==1.8.10
pycrypto==2.6
pygame==1.9.1
pygobject==2.28.3
pygoocanvas==0.14.2
pygtk==2.24.0
pygtksourceview==2.10.1
pyHook==1.5.1
pyopengl==3.0.2b2
pyparsing==2.0.2
pyrsvg==2.32.1
pyrtlsdr==0.2.0
pyserial==2.7
python-dateutil==2.4.2
python-tesseract==0.9
pytify==0.1b0
PyUserInput==0.1.9
pywin32==218
PyYAML==3.10
relatorio==0.6.1
requests==2.7.0
six==1.9.0
tiger==0.3
virtualenv==1.11.6
websocket-client==0.11.0
```

Main process seems to be doing all the work, while multiprocessing forks are idling.
[image]

EarToEarOak, 2015-09-15:
I replaced multiprocessing with threading in 0c89734854023a8ca4e1302ab8fa381a402d52b5 4 days ago.  There are a handful of optimisations and smaller issues which I want to fix before I build the stand-alone binaries, but that shouldn't take too long.

The main problem is drawing the plot has to occur on the main thread which can take some time with large datasets.  To reduce the hangs try turning off real time plotting or switching the display mode to 'Status' whilst scanning.  If you still want to see the plot as you scan you can set the display to 'Preview' but this requires visvis which due to some problems isn't available with the stand-alone executables for the moment.

leitao, 2016-05-29:
Thanks for the workaround, but it is not working here.
I am using Debian jessie on amd64, and after installing visvis using pip, I now see " Click for preview"  (instead of "visvis is not installed" ) and I even can drag and move it, but  I can't see the preview.

Even after stopping the scan, I am not able to see the data, since " Click for preview"  is visible all the time and does not get away.

EarToEarOak, 2016-06-04:
@leitao does a second 'Preview' window appear with the plot?

closed 2018-04-13 by EarToEarOak