EarToEarOak/RTLSDR-Scanner #17: Ubuntu 14.04 compilation error

MagicianT, 2016-02-05:
running install
running build
running build_py
running install_lib
byte-compiling /usr/local/lib/python2.7/dist-packages/rtlsdr/rtlsdraio.py to rtlsdraio.pyc
  File "/usr/local/lib/python2.7/dist-packages/rtlsdr/rtlsdraio.py", line 15
    def __init__(self, func_start, func_stop=None, queue_size=20, *, loop=None):
                                                                   ^
SyntaxError: invalid syntax

running install_egg_info
Removing /usr/local/lib/python2.7/dist-packages/pyrtlsdr-0.1.1.egg-info
Writing /usr/local/lib/python2.7/dist-packages/pyrtlsdr-0.1.1.egg-info

EarToEarOak, 2016-02-10:
This is an error from the [pyrtlsdr](https://github.com/roger-/pyrtlsdr) library but shouldn't cause an issue with the installation and running.
I believe it's to do with rtlsdraio.py not supporting Python 2.x

Does the software still function despite this?

EarToEarOak, 2016-04-16:
The issue has been fixed in recent commits to pyrtlsdr.

closed 2016-04-16 by EarToEarOak