EarToEarOak/RTLSDR-Scanner #20: Scan hangs during the ~5% of the scan (Windows)

Alex131089, 2016-03-25:
Maybe related to #13, but for me the scan hangs a short time after the scan starts in latest versions.
Latest working version is v1.0.21972.32419. I tried to let it run a bit, with the display mode to "Status", but it seems to run forever.

On some of my tries I had a PLL lock error message.

Env : Win8, NooElec Mini 2+

EarToEarOak, 2016-04-05:
Do you know which version of matplotlib you are using?

Alex131089, 2016-04-05:
I'm using the standalone version (I should have tell this before), so the one included.

EarToEarOak, 2016-04-13:
Does the command 'rtl_test' or any other program (such as SDR#) work without problems?

Alex131089, 2016-04-13:
No big problem with SDR# (sdr_test not tested) and with version 1.0.21972.32419 it's ok

leitao, 2016-05-29:
I have the very same problem here. After a while, on long frequency scans, the window hangs and I am not able to recover.

closed 2018-04-13 by EarToEarOak