EarToEarOak/RTLSDR-Scanner #21: RTLSDR-Scanner segfaulting

kornos, 2016-04-19:
I downloaded today 19 Apr RTLSDRSCANNER from github , but it doesn't work for me when i try to scan frequency range 900Mhz-1Ghz. 
It keeps segfaulting:

Found Elonics E4000 tuner
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
[E4K] PLL not locked for 969000000 Hz!
libusb: error [submit_control_transfer] submiturb failed error -1 errno=9
rtlsdr_demod_write_reg failed with -1
Segmentation fault (core dumped)

or

Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
[E4K] PLL not locked for 905250000 Hz!
Segmentation fault (core dumped)
kornos@ubuntu:~/projects/SDR/RTL

Dmesg output:
[165794.769721] python[52465]: segfault at e0 ip 00007fdde4e23c84 sp 00007fddcce88000 error 4 in libpthread-2.21.so[7fdde4e1a000+18000]
[165857.194711] python[52566]: segfault at e0 ip 00007fa429c77c84 sp 00007fa4054d3000 error 4 in libpthread-2.21.so[7fa429c6e000+18000]
[166100.351597] python[52765]: segfault at 30 ip 00007f1fb6e41250 sp 00007f1f97ffdff0 error 4 in libusb-1.0.so.0.1.0[7f1fb6e39000+17000]
[166342.513662] python[53132]: segfault at 1000000f4 ip 00007f7641811c84 sp 00007f761fffe000 error 4 in libpthread-2.21.so[7f7641808000+18000]

I have Ubuntu 14.10 configured properly , kernel 4.4.6 , eveything else(SDFR related)  is working OK(gqrx,etc).
When I scan the range with implicit values(87-108 Mhz), it works ok.

EarToEarOak, 2016-04-22:
Have you got a dongle with a different tuner to the E4000 that you can try?  I'm wondering if it has something to do with the frequency gap the E4000 has at about 1GHz.

kornos, 2016-04-22:
Hello AL

Unfortunately I don't have another tuner.

But I scanned the same frequency ranges with the Windows software named
Touchstone and
it worked like a charm.

E4000 frequency gap is between 1100-1250 Mhz,I scanned 900 Mhz-1Ghz.

Right now  I've tried scanning the frequency range 1700-1800 Mhz, it
segfaults as well.

IT SEEMS that segfault occurs imediately after *PLL not locked* message:
[E4K] PLL not locked for 1735000000 Hz!

This happened also with 900-1000 Mhz scan.
Maybe RTLSDRScanner is not taking into accound this *PLL not locked *
condition errors?

Thanks
Victor

kornos, 2016-04-22:
Unfortunately I don't have another tuner.

But I scanned the same frequency ranges with the Windows software named Touchstone and
it worked like a charm.

E4000 frequency gap is between 1100-1250 Mhz,I scanned 900 Mhz-1Ghz.

Right now  I've tried scanning the frequency range 1700-1800 Mhz, it segfaults as well.

IT SEEMS that segfault occurs imediately after *PLL not locked* message:
[E4K] PLL not locked for 1735000000 Hz!

This happened also with 900-1000 Mhz scan.
Maybe RTLSDRScanner is not taking into accound this *PLL not locked * condition errors?

Thanks
Victor

EarToEarOak, 2016-04-22:
The errors you are getting point to a USB transfer error or similar which is causing the driver to crash.

Could you try it again, setting the dwell to the lowest?

Thanks

kornos, 2016-04-22:
Ok , i tried it with dwell = 8 ms(minimum value).

With gain=-1 :
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
[E4K] PLL not locked for 899250000 Hz!
Segmentation fault (core dumped)

With gain=9:
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
[E4K] PLL not locked for 89000000 Hz!
libusb: error [submit_control_transfer] submiturb failed error -1 errno=9
rtlsdr_demod_write_reg failed with -1
Segmentation fault (core dumped)

Sometimes it segfaults , sometime it just hangs , sometimes i can kill the
window , sometimes not(i heve to kill the process).

From 10 tries , it finished scanning only ONCE! (it's a first , it didn't
happen before).

Some strace dump of python process:

recvmsg(3, 0x7ffcf75d0d40, 0)           = -1 EAGAIN (Resource temporarily
unavailable)
poll([{fd=3, events=POLLIN}, {fd=4, events=POLLIN}], 2, 498) = 1 ([{fd=3,
revents=POLLIN}])
recvmsg(3, {msg_name(0)=NULL,
msg_iov(1)=[{"\6\0\300I\345Y\216\20%\2\0\0p\1\300\3\0\0\0\0\335\1s\3\10\0\275\2\20\0\1\0",
4096}], msg_controllen=0, msg_flags=0}, 0) = 32
recvmsg(3, 0x7ffcf75d0d60, 0)           = -1 EAGAIN (Resource temporarily
unavailable)
poll([{fd=3, events=POLLIN|POLLOUT}], 1, 4294967295) = 1 ([{fd=3,
revents=POLLOUT}])
writev(3, [{"&\30\2\0p\1\300\3", 8}, {NULL, 0}, {"", 0}], 3) = 8
poll([{fd=3, events=POLLIN}], 1, 4294967295) = 1 ([{fd=3, revents=POLLIN}])
recvmsg(3, {msg_name(0)=NULL,
msg_iov(1)=[{"\1\1\301I\0\0\0\0%\2\0\0\0\0\0\0\335\1s\3\10\0\275\2\20\0\0\0\0\0\0\0",
4096}], msg_controllen=0, msg_flags=0}, 0) = 32
recvmsg(3, 0x7ffcf75d0510, 0)           = -1 EAGAIN (Resource temporarily
unavailable)
recvmsg(3, 0x7ffcf75d0510, 0)           = -1 EAGAIN (Resource temporarily
unavailable)

Thanks
Victor

kornos, 2016-04-22:
Ok , i tried it with dwell = 8 ms(minimum value).

With gain=-1 :
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
[E4K] PLL not locked for 899250000 Hz!
Segmentation fault (core dumped)

With gain=9:
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
[E4K] PLL not locked for 89000000 Hz!
libusb: error [submit_control_transfer] submiturb failed error -1 errno=9
rtlsdr_demod_write_reg failed with -1
Segmentation fault (core dumped)

Sometimes it segfaults , sometime it just hangs , sometimes i can kill the window , sometimes not(i heve to kill the process).

From 10 tries , it finished scanning only ONCE! (it's a first , it didn't happen before).

Some strace dump of python process:

recvmsg(3, 0x7ffcf75d0d40, 0)           = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=3, events=POLLIN}, {fd=4, events=POLLIN}], 2, 498) = 1 ([{fd=3, revents=POLLIN}])
recvmsg(3, {msg_name(0)=NULL, msg_iov(1)=[{"\6\0\300I\345Y\216\20%\2\0\0p\1\300\3\0\0\0\0\335\1s\3\10\0\275\2\20\0\1\0", 4096}], msg_controllen=0, msg_flags=0}, 0) = 32
recvmsg(3, 0x7ffcf75d0d60, 0)           = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=3, events=POLLIN|POLLOUT}], 1, 4294967295) = 1 ([{fd=3, revents=POLLOUT}])
writev(3, [{"&\30\2\0p\1\300\3", 8}, {NULL, 0}, {"", 0}], 3) = 8
poll([{fd=3, events=POLLIN}], 1, 4294967295) = 1 ([{fd=3, revents=POLLIN}])
recvmsg(3, {msg_name(0)=NULL, msg_iov(1)=[{"\1\1\301I\0\0\0\0%\2\0\0\0\0\0\0\335\1s\3\10\0\275\2\20\0\0\0\0\0\0\0", 4096}], msg_controllen=0, msg_flags=0}, 0) = 32
recvmsg(3, 0x7ffcf75d0510, 0)           = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(3, 0x7ffcf75d0510, 0)           = -1 EAGAIN (Resource temporarily unavailable)

kornos, 2016-04-22:
Ok AL , some good news , I think I started tracking down the problem.

The problem is on rtl_scan function:
I modfied scan.py a little bit , now it doesn't segfault anymore *THE FIRST
TIME I'm running it with the minimum dwell(8ms):

def rtl_scan(self, freq):
        try:
            self.sdr.set_center_freq(freq + self.lo)
        except OSError as error:
            post_event(self.notify, EventThread(Event.ERROR,
                                                0, error))
            capture = []
            return capture

All the problems are about set_center_freq() , PLL not locked error and
Python threading model.
I suspect also some libusb bug , but i have to compile it with debug
symbols and see wqhere it crashes.

So , for short now the situation looks like this:
1. With my modification , program works JUST THE FIRST TIME YOU RUN IT EACH
TIME with min dwell value (though the progress bar does not reach 100%)
When it encounters PLL not locked error , it just returs and draw graph -
this is because of me not knowing Python and RTLSDRScanner code source yet.
2. When you run 2nd,3rd,4th,etc times and program with min dwell values it
just prints a lot of :
ibusb: warning [add_to_flying_list] failed to arm first timerfd (errno 9)
rtlsdr_demod_read_reg failed with -99
libusb: warning [add_to_flying_list] failed to arm first timerfd (errno 9)
3. When you run the program (even for the 1st time) with bigger dwell
values , hell breaks loose: all kind of weird errors like this one below
[E4K] PLL not locked for 946750000 Hz!
python: ../nptl/pthread_mutex_lock.c:355: __pthread_mutex_lock_full:
Assertion `(-(e)) != 3 || !robust' failed.

or

Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
[E4K] PLL not locked for 945500000 Hz!
libusb: error [submit_control_transfer] submiturb failed error -1 errno=9
rtlsdr_demod_write_reg failed with -1
libusb: error [submit_control_transfer] submiturb failed error -1 errno=9
rtlsdr_demod_read_reg failed with -1

I'm pretty sure that the source of all this errors is the fact that *PLL
not locked* error condition is not properly taken care of.
(now I don't know at each layer: RTLSDRSCANNER .pyrtlsdr or libusb).

I scan.y I replaced the exception WindowsError with OSError , to run
correctly on Linux also.
I attach scan.py .

Thanks Victor

2016-04-22 20:31 GMT+03:00 Punk Rider <tmucbvest@gmail.com>:

kornos, 2016-04-22:
Ok AL , some good news , I think I started tracking down the problem.

The problem is on rtl_scan function:
I modfied scan.py a little bit , now it doesn't segfault anymore *THE FIRST TIME I'm running it with the minimum dwell(8ms):

def rtl_scan(self, freq):
        try:
            self.sdr.set_center_freq(freq + self.lo)
        except OSError as error:
            post_event(self.notify, EventThread(Event.ERROR,
                                                0, error))
            capture = []
            return capture

All the problems are about set_center_freq() , PLL not locked error and Python threading model.
I suspect also some libusb bug , but i have to compile it with debug symbols and see where it crashes.

So , for short now the situation looks like this:
1. With my modification , program works JUST THE FIRST TIME YOU RUN IT EACH TIME with min dwell value (though the progress bar does not reach 100%)
When it encounters PLL not locked error , it just returns and draw graph - this is because of me not knowing Python and RTLSDRScanner code source yet.
2. When you run 2nd,3rd,4th,etc times and program with min dwell values it just prints a lot of :
ibusb: warning [add_to_flying_list] failed to arm first timerfd (errno 9)
rtlsdr_demod_read_reg failed with -99
libusb: warning [add_to_flying_list] failed to arm first timerfd (errno 9)
3. When you run the program (even for the 1st time) with bigger dwell values , hell breaks loose: all kind of weird errors like this one below
[E4K] PLL not locked for 946750000 Hz!
python: ../nptl/pthread_mutex_lock.c:355: __pthread_mutex_lock_full: Assertion `(-(e)) != 3 || !robust' failed.

or

Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
[E4K] PLL not locked for 945500000 Hz!
libusb: error [submit_control_transfer] submiturb failed error -1 errno=9
rtlsdr_demod_write_reg failed with -1
libusb: error [submit_control_transfer] submiturb failed error -1 errno=9
rtlsdr_demod_read_reg failed with -1

I'm pretty sure that the source of all this errors is the fact that *PLL not locked* error condition is not properly taken care of.
(now I don't know at each layer: RTLSDRSCANNER ,pyrtlsdr or libusb).

In scan.py I replaced the exception WindowsError with OSError , to run correctly on Linux also.
I attach scan.py .
[scan.py.zip](https://github.com/EarToEarOak/RTLSDR-Scanner/files/232441/scan.py.zip)

kornos, 2016-04-22:
Well , i suspect that the problem is likely to be in librtlsdr library:

https://github.com/roger-/pyrtlsdr/issues/22

kornos, 2016-04-22:
After browsing for related problem on the Internet , I think the problem is
likely to be in librtlsdr :

https://github.com/roger-/pyrtlsdr/issues/22

2016-04-22 22:08 GMT+03:00 Punk Rider <tmucbvest@gmail.com>:

EarToEarOak, 2016-04-27:
It's a shame that this isn't exposed but I'm not sure that it relates to your issue.

Looking at the driver this is only a warning that it wasn't able to tune to the requested frequency, in the driver there is nothing that stops the USB transfer as you are experiencing.

I've run it for over 8 hours continuously on a Ubuntu 14.10 system with a E4000 and seen quite a few of these warnings but no transfer errors and a few hours each with a FC0012, R820T and R820T2.  It looks like the RTL2832U still keeps transferring despite these warnings.

I'll keep looking...

kornos, 2016-04-28:
Hello Al

I have some news to report after further testing.
I finally got RTLSDRScanner not to segfault anymore.

The problem seemed to be that I installed pyrtlsdr via pipi
command.Unfortunately pyrtlsdr from pip command is flawed:

kornos@ubuntu:~/projects/SDR/pyrtlsdr$ sudo pip install pyrtlsdr
Downloading/unpacking pyrtlsdr
  Downloading pyrtlsdr-0.2.2-py2-none-any.whl
Installing collected packages: pyrtlsdr
Compiling /tmp/pip-build-1mYngd/pyrtlsdr/rtlsdr/rtlsdraio.py ...
  File "/tmp/pip-build-1mYngd/pyrtlsdr/rtlsdr/rtlsdraio.py", line 24
    def __init__(self, func_start, func_stop=None, queue_size=20, *,
loop=None):
                                                                   ^
SyntaxError: invalid syntax

Successfully installed pyrtlsdr
Cleaning up...

So I've uninstalled pyrtlsdr with pip unistalled and after that I installed
pyrtlsdr from github with: sudo python setup.py install

Now RTLSDRScanner is not segfaulting anymore , instead it displays an error
message(I attached the screenshots).
When PLL not locked appears , the program shows this error message , it
plots with matlib the spectrogram *UP* to that frequency of PLL not locked
(and no further) and it stops.
It would be nice if the program could continue after this error(eventually
resetting the device or something like this).

Thanks
Victor

kornos, 2016-04-28:
[image]
[image]
[image]
[image]
[image]
[image]

I have some news to report after further testing.
I finally got RTLSDRScanner not to segfault anymore.

The problem seemed to be that I installed pyrtlsdr via pipi command.Unfortunately pyrtlsdr from pip command is flawed:

kornos@ubuntu:~/projects/SDR/pyrtlsdr$ sudo pip install pyrtlsdr
Downloading/unpacking pyrtlsdr
  Downloading pyrtlsdr-0.2.2-py2-none-any.whl
Installing collected packages: pyrtlsdr
Compiling /tmp/pip-build-1mYngd/pyrtlsdr/rtlsdr/rtlsdraio.py ...
  File "/tmp/pip-build-1mYngd/pyrtlsdr/rtlsdr/rtlsdraio.py", line 24
    def __init__(self, func_start, func_stop=None, queue_size=20, *, loop=None):
                                                                   ^
SyntaxError: invalid syntax

Successfully installed pyrtlsdr
Cleaning up...

So I've uninstalled pyrtlsdr with pip unistalled and after that I installed pyrtlsdr from github with: sudo python setup.py install

Now RTLSDRScanner is not segfaulting anymore , instead it displays an error message(I attached the screenshots).
When PLL not locked appears , the program shows this error message , it plots with matlib the spectrogram *UP* to that frequency of PLL not locked
(and no further) and it stops.
It would be nice if the program could continue after this error(eventually resetting the device or something like this).

Thanks
Victor

kornos, 2016-04-28:
[image]
[image]
[image]
[image]
[image]
[image]

I have some news to report after further testing.
I finally got RTLSDRScanner not to segfault anymore.

The problem seemed to be that I installed pyrtlsdr via pipi command.Unfortunately pyrtlsdr from pip command is flawed:

kornos@ubuntu:~/projects/SDR/pyrtlsdr$ sudo pip install pyrtlsdr
Downloading/unpacking pyrtlsdr
  Downloading pyrtlsdr-0.2.2-py2-none-any.whl
Installing collected packages: pyrtlsdr
Compiling /tmp/pip-build-1mYngd/pyrtlsdr/rtlsdr/rtlsdraio.py ...
  File "/tmp/pip-build-1mYngd/pyrtlsdr/rtlsdr/rtlsdraio.py", line 24
    def __init__(self, func_start, func_stop=None, queue_size=20, *, loop=None):
                                                                   ^
SyntaxError: invalid syntax

Successfully installed pyrtlsdr
Cleaning up...

So I've uninstalled pyrtlsdr with pip unistalled and after that I installed pyrtlsdr from github with: sudo python setup.py install

Now RTLSDRScanner is not segfaulting anymore , instead it displays an error message(I attached the screenshots).
When PLL not locked appears , the program shows this error message , it plots with matlib the spectrogram *UP* to that frequency of PLL not locked
(and no further) and it stops.
It would be nice if the program could continue after this error(eventually resetting the device or something like this).

Thanks
Victor
[image]
[image]
[image]

kornos, 2016-04-28:
Yep , the guys from pyrtlsdr confirmed a bug in pip install pyrtlsdr procedure.
They are fixing it now.

https://github.com/roger-/pyrtlsdr/issues/42

closed 2016-06-04 by EarToEarOak