EarToEarOak/RTLSDR-Scanner #22: Windows 10 Standalone - import error for librtlsdr

tpainter, 2016-06-20:
The stand alone executable doesn't seem to work on Windows 10 (used v1.0.22298.18049 release download.)

The error that is displayed in the window is:
`Import error: Error loading librtlsdr. Make sure that librtlsdr (and all of its dependencies) are in your path`

`Error importing libraries`

The proper drivers are installed and working, e.g. SDR# works fine.

EarToEarOak, 2016-06-20:
Try downloading the [RTLSDR driver](http://sdr.osmocom.org/trac/attachment/wiki/rtl-sdr/RelWithDebInfo.zip) and make sure it's in your path.

EarToEarOak, 2016-06-22:
Another thought: make sure you use the files from the x64 directory for stand alone version.

tpainter, 2016-06-24:
Seems like the issue was x64 vs x32. It is working now.

closed 2016-06-24 by tpainter

EarToEarOak, 2016-06-24:
Glad to hear its' working.  I'll get round to updating the documentation soon to make it clearer.