EarToEarOak/RTLSDR-Scanner #24: RFE: Linux tweaks for easier distro packaging

yarda, 2016-07-14:
Hi,

I have just packaged RTLDR-Scanner into Fedora [1]. It turned out few things could be improved for the package to be installed system-wide. I think this could ease packaging for other distros as well. I am attaching distro specific patch I used in Fedora. I am pretty sure it could be done more elegant and generic way - it's just for the insight and not currently ready for PR. What the patch is intended for:

- handling of modularization of the code - modules are installed to the python_sitelib/rtlsdr-scanner (e.g. /usr/lib/python2.7/site-packages/rtlsdr-scanner) and main executable to the /usr/bin (/usr/bin/rtlsdr_scan). The patch currently uses dirty hack for this to work. I think it may be enough to just create subdirectory, e.g. rltsdr-scanner and import from it. In such way it wouldn't matter where the distro put this directory while it stays in the python path.
- handling of resources installed to the /usr/share/rtlsdr-scanner/res - it may use some prefix which should be easily set by distro
- reading icon from the /usr/share/icons/hicolor/256x256/apps in the PNG format (the same as used in the desktop file, I will PR the desktop file and appdata file later) - the .ico is not optimal for Linux and using /usr/share/icons is much more appropriate, so it could check the path in the code conditionaly.

Maybe I could also try to implement these ideas better way and PR later.

[1] https://bugzilla.redhat.com/show_bug.cgi?id=1356048
[patch.zip](https://github.com/EarToEarOak/RTLSDR-Scanner/files/364014/patch.zip)

EarToEarOak, 2016-07-18:
Thanks for that, I'll start looking into when I get chance.

I didn't go down this route myself due to the extra work getting the stand alone version to play nicely.

drws, 2016-08-25:
This is an example working .desktop file:

    [Desktop Entry]
    Version=1.0
    Type=Application
    Name=RTLSDR Scanner
    Comment=A cross platform Python frequency scanning GUI for the OsmoSDR rtl-sdr library
    Exec=rtlsdr_scan
    Icon=/opt/rtlsdr-scanner/rtlsdr_scan.ico
    Categories=HamRadio;

Only the icon path is hardcoded, because `/usr/share/icons/` is not yet used (included in this bug report). Also, #25 could be marked as a duplicate of this one?

yarda, 2016-08-25:
Thanks. I think you should use /usr/share/icons (or some variable there replaceable during the install phase), but it's no problem for downstream to patch it. I think #25 is duplicate then.

EarToEarOak, 2017-01-15:
I've added your example desktop file in 1d7565fc7eb70c154c8128d1e98dc67ce25617a7, although I've removed the icon key as this should be modified by the package maintainer.

Thanks again.

closed 2017-01-15 by EarToEarOak