EarToEarOak/RTLSDR-Scanner #25: Desktop file for Linux

yarda, 2016-07-14:
I am attaching desktop file for Linux. Currently it suppose that the main executable is located in the /usr/bin. Maybe it could use some variable and substitute the path to be more generic. I am not sure where you want to add it, so I am attaching it and not doing PR, but I could do PR if you prefer it.

[desktop-file.zip](https://github.com/EarToEarOak/RTLSDR-Scanner/files/364045/desktop-file.zip)

EarToEarOak, 2016-07-18:
Thank you

I'll bear this in mind while considering #24

drws, 2016-08-25:
@EarToEarOak please also consider this corrected .desktop file, the first one has a few errors in it:

    [Desktop Entry]
    Version=1.0
    Type=Application
    Name=RTLSDR Scanner
    Comment=A cross platform Python frequency scanning GUI for the OsmoSDR rtl-sdr library
    Exec=rtlsdr_scan
    Icon=/opt/rtlsdr-scanner/rtlsdr_scan.ico
    Categories=HamRadio;

EarToEarOak, 2016-08-30:
Duplicate of #24

closed 2016-08-30 by EarToEarOak