EarToEarOak/RTLSDR-Scanner #27: AttributeError: 'RtlSdr' object has no attribute 'valid_gains_db'

pschonmann, 2016-07-22:
Trying to run scanner from latest github code. Giving that error. Running on kali linux rolling updates.

Linux kali 4.6.0-kali1-amd64 #1 SMP Debian 4.6.4-1kali1 (2016-07-21) x86_64 GNU/Linux

root@kali:~/sdr/RTLSDR-Scanner/src# ./rtlsdr_scan.py 
RTLSDR Scanner

Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Traceback (most recent call last):
  File "./rtlsdr_scan.py", line 129, in <module>
    frame = FrameMain(APP_NAME)
  File "/root/sdr/RTLSDR-Scanner/src/main_window.py", line 149, in __init__
    self.devicesRtl = get_devices_rtl(self.settings.devicesRtl)
  File "/root/sdr/RTLSDR-Scanner/src/devices.py", line 122, in get_devices_rtl
    device.gains = sdr.valid_gains_db
AttributeError: 'RtlSdr' object has no attribute 'valid_gains_db'

EarToEarOak, 2016-07-22:
Try updating [pyrtlsdr](https://github.com/roger-/pyrtlsdr), older versions don't have the 'valid_gains_db' method.

pschonmann, 2016-07-22:
Thanks, magic happened !
pip install --upgrade pyrtlsdr works

JiangXL, 2016-07-22:
@pschonmann   Thanks,it also works for me

closed 2016-07-24 by EarToEarOak