EarToEarOak/RTLSDR-Scanner #35: Install fails

Dougga, 2017-04-20:
The installation script fails multiple times on Win7 x64.
1. The wxPython install fails.  This can be mitigated by searching the internet for the appropriate downloadable installation program and installing manually.  [begs the question why the install script exists]
2. After the above, the installation script gets a bit further before failing.  The final piece is the installation of the scanner.  I get a DOS Box opening up with a script that errors many times indicating something about proxy authentication failure.  I have configured and reconfigured this machine to not have any proxies and I still get these proxy errors.  I assume this is some hack for an installation script, but it's a mess. 

 Could you please fix the thing: installshield installation app.  If you're a linux shop, someone in India will do it for your for $50 I'm sure.

Dougga, 2017-04-20:
It appears the script has the proxy hard coded for whatever environment it was developed in.

Dougga, 2017-04-20:
Found what might be the problem... an intrusion protection alert

protection alert" action="alert" reason="FILE-OTHER RealNetworks RealPlayer compressed skin overflow attempt" group="340" srcip="144.76.43.76" dstip="10.1.2.3" proto="6" srcport="80" dstport="57520" sid="21420" class="Attempted User Privilege Gain" priority="1" generator="1" msgid="0"

EarToEarOak, 2017-04-21:
The [install script](https://github.com/EarToEarOak/RTLSDR-Scanner/blob/master/nsis/rtlsdr_scanner.nsi) doesn't have a hard coded proxy, it uses the system setting found in Control Panel -> Internet Options -> Connections.

From the intrusion detection log it seems to be a false positive;  the log refers to the RTLSDR driver for windows which according to [VirusTotal](https://virustotal.com/en/file/7652fc6602fd515436faca63b018117fa5dd9df60de18f11e77ccf3f461ede50/analysis/1492206393/) is clean,  the same is also true of the [wxPython](https://virustotal.com/en/file/3ac291535bcf15fd5a15bcd29b066570e8d2a0cab4f3b92a2372f41aa09a4f48/analysis/1492550291/) installation file.

 I wonder if the error you are seeing is a generic one due to your security software blocking the download, but it's hard to say without seeing the actual error.

Optimustron, 2017-08-25:
I can't install the 64-bit program, however I installed the 32-bit version. The 64-bit gives this message:
**Import error: Error loading librtlsdr. Make sure librtlsdr <and all of its dependencies> are in your path
Error importing libraries
Press [Return] to exit**
When you hit [Return] then appears:
**Traceback <most recent call last>:
  File "__main__.py", line 41, in <module>
NameError: name 'exit' is not defined
Failed to execute script __name__**

kidoda, 2017-08-26:
@Dougga I noticed you said: "This can be mitigated by searching the internet for the appropriate downloadable installation program and installing manually." Is the installation program you downloaded from an official source? There's a ton of adware out in the web that they change the filename to whatever you search for so you download it instead.

Dougga, 2017-08-28:
At this point I don't recall, but to say I'm conservative with installation sources would be an understatement.  I'm fairly confident I found a solid source.  I may try to retrace my steps.

EarToEarOak, 2017-08-31:
I only release the installers on [GitHub](https://github.com/EarToEarOak/RTLSDR-Scanner/releases/latest).

I think the error you are seeing is because the 64 bit versions require the 64 bit version of the rtlsdr driver to be in your path.

JM1XTB, 2017-10-06:
I tried 64-bit program but failed. The installation log is following.

C:\RTL-SDR>rtlsdr_scan-windows-64bit.exe
Traceback (most recent call last):
  File "__main__.py", line 33, in <module>
  File "c:\python27\Lib\site-packages\PyInstaller\loader\pyimod03_importers.py",
 line 389, in load_module
  File "site-packages\matplotlib\__init__.py", line 1138, in <module>
  File "site-packages\matplotlib\__init__.py", line 971, in rc_params
  File "site-packages\matplotlib\__init__.py", line 800, in matplotlib_fname
  File "site-packages\matplotlib\__init__.py", line 655, in _get_configdir
  File "site-packages\matplotlib\__init__.py", line 609, in _get_config_or_cache_dir
  File "site-packages\matplotlib\__init__.py", line 218, in _is_writable_dir
UnicodeDecodeError: 'ascii' codec can't decode byte 0x90 in position 9: ordinal
not in range(128)
Failed to execute script __main__

Can I escape this failure ? Please let me know.

JM1XTB, 2017-10-06:
Above **main** is "__main__" same as Optimustron-san's screen shot.

EarToEarOak, 2017-11-10:
Have you had any luck with this?  I'm still unable to replicate the error.

closed 2018-04-13 by EarToEarOak