EarToEarOak/RTLSDR-Scanner #38: Problem with multiple RTL-SDR sticks

gvanem, 2018-01-30:
I have 2 RTL-SDR sticks installed on my Win-10 PC. The `rtlsdr_scanner` worked the 1st time I ran it. But on a subsequent run Python throws this error when I run `python2 -m rtlsdr_scanner  -i 1`:
```
RTLSDR Scanner

usb_open error -12/LIBUSB_ERROR_NOT_SUPPORTED
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Traceback (most recent call last):
  File "F:\ProgramFiler\Python27\lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "F:\ProgramFiler\Python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "F:\gv\dx-radio\RTLSDR-Scan\rtlsdr_scanner\__main__.py", line 126, in <module>
    frame = FrameMain(APP_NAME)
  File "rtlsdr_scanner\main_window.py", line 177, in __init__
    self.__create_toolbars()
  File "rtlsdr_scanner\main_window.py", line 201, in __create_toolbars
    self.remoteControl)
  File "rtlsdr_scanner\panels.py", line 114, in __init__
    self.create_plot()
  File "rtlsdr_scanner\panels.py", line 316, in create_plot
    self.set_plot_title()
  File "rtlsdr_scanner\panels.py", line 375, in set_plot_title
    gain = self.settings.devicesRtl[self.settings.indexRtl].gain
IndexError: list index out of range
```
The only success I have with my setup is this method:
```
f:\gv\dx-radio\RTLSDR-Scan> start rtl_tcp -d2
f:\gv\dx-radio\RTLSDR-Scan> python2 -m rtlsdr_scanner  --remote 127.0.0.1:1234
```
Is there some config-file that sets the `rtl_tcp` parameters somewhere? Seems to be that which is messing up.

EarToEarOak, 2018-02-02:
Does sound like a configuration problem, until I find a fix you could try deleting the device settings in the registry at:
HCU\Software\rtlsdr-scanner\DevicesRTL

gvanem, 2018-02-03:
You mean the `REG_DWORD` value `HCU\Software\rtlsdr-scanner\DevicesRTL\isDevice`? That worked.

EarToEarOak, 2018-02-05:
Good, looks like it got confused with local vs. remote.

closed 2018-02-05 by EarToEarOak