EarToEarOak/RTLSDR-Scanner #43: device permissions

Daddio61, 2018-05-05:
On running in Windows 10 1803 Build 17134.1 I get the following response from Python 27\\python.exe

Found Rafael Micro R820T tuner
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules

And when clicking on START in the GUI, Message box ERROR code -1 when opening SDR (device index =0)

I ran BBCR2.rfs after closing SDR# V1.0.0.1666, received the Python error, unplugged and reinserted the RTL SDR V3 then got the same message and then clicked START on the GUI to achieve the above repeatable error.

Great program thanks for coding it!!

EarToEarOak, 2018-05-05:
What does the output of `rtl_test `give?

Daddio61, 2018-05-05:
Found 1 device(s):
     0:    Realtek, RTL_SDR_V3, SN: RTL_SDR_V3

Using device 0: Generic RTL2382U OEM
Found Rafael Micro R820T tuner
Supported gain values (29):  0.0 0.9 1.4  etc
Sampling at 2048000 S/s

Info: this tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
lost at least 136 bytes
lost at least 84 bytes
lost at least 5s bytes

ad nauseum!!

EarToEarOak, 2018-05-07:
It looks like your device is working and setup properly (you might get fewer lost samples if you use a different USB port, YMMV).
The error you see is usually due to a driver installation problem or another driver or software using the device but the `rtl_test` output shows it's functioning properly.
I'll perform some more testing on Windows 10 and report back.

dspboydsp, 2018-08-05:
I am having issues.  Can you help?  When I run rtl_test I get:   
Found 2 device(s):
  0:   , , SN:
  1:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
usb_open error -12
Failed to open rtlsdr device #0.

EarToEarOak, 2018-08-14:
That sounds like the Windows TV driver for the dongle is installed.  Uninstall it first.

closed 2018-08-14 by EarToEarOak

rudi48, 2019-02-07:
@EarToEarOak
I also have this error: usb_open error -12
under Windows 10 64 Bit.
The program SDRsharp.exe does work, RTLSDR-Scanner starts, but shows the I/O error.
I have read yout hint with the "Windows TV driver" but I cannot find it.
How can I uninstall that driver?
Regards, Rudolf
[image]

rudi48, 2019-02-07:
@EarToEarOak
I finally found the solution for "error -12":
https://forums.radioreference.com/threads/fix-for-usb-open-error-12-in-libusb-dll-applications.354595/
With SDR plugged in...Device Manager, right-click device, Uninstall, check box to remove drivers)
 Run the Zadig installer, checked the entry for "Options->List All Devices", then cleared the check for "Ignore Hubs or Composite Devices". I then located the RTLSDR???? (Composite) entry and installed the driver for this device only. Zadig installer prompted me that I was replacing a system driver, I accepted and closed the Zadig installer.
[image]

sysfu, 2019-08-21:
Thank you @rudi48! 

The workaround you posted resolved the problem for me as well.

Daddio61, 2020-07-15:
> On running in Windows 10 1803 Build 17134.1 I get the following response from Python 27\python.exe
>
> Found Rafael Micro R820T tuner
> usb_open error -3
> Please fix the device permissions, e.g. by [...]

> It looks like your device is working and setup properly (you might get fewer lost samples if you use a different USB port, YMMV).
> The error you see is usually due to a driver installation problem or [...]

Hi did you find a solution to my issue...I'm still having it!