EarToEarOak/RTLSDR-Scanner #50: Python 3 Support

Evidlo, 2018-09-24:
I worked a bit on adding [Python3 and wxPython4 support](https://github.com/Evidlo/RTLSDR-Scanner/tree/python3).  If you make a `python3` branch, I'll submit a PR so others can test it.

rhertzog, 2019-08-15:
Python 2.x will no longer be supported by their upstream developers in 2020. Thus Debian developers are actively removing Python 2 support in Debian Testing with the goal of getting rid of Python 2 in Debian 11 (bullseye). This is starting to affect Kali Linux and RTLSDR-Scanner as a package in Kali (right now, python-visvis was broken by the removal from Debian of python-imageio). I expect more problems to show up in the future. I note that "visvis" is already ported to Python 3 so this shouldn't be an issue for you.

This is also tracked on the Kali side here: https://gitlab.com/kalilinux/packages/rtlsdr-scanner/issues/1

@EarToEarOak What are your plans for Python 3 support ?

@Evidlo I think you can submit your merge request again the master branch, no? Even if it doesn't get merged immediately, at least your code is available for others...

rhertzog, 2019-08-15:
Ah I see you already submitted the merge request here: https://github.com/EarToEarOak/RTLSDR-Scanner/pull/51

yarda, 2019-11-07:
I introduced #51 into Fedora downstream (Rawhide / f32) otherwise they would drop rtlsdr-scanner package from Fedora, because Fedora is dropping Python 2.