EarToEarOak/RTLSDR-Scanner #54: Installation on MacOS High Sierra 10.13.6 and Mojave 10.14.3

rudi48, 2019-02-07:
Tested on MacOS High Sierra 10.13.6 and Mojave 10.14.3
Xcode must be installed.
In the Terminal:
# Install Homebrew: https://brew.sh/
# Xcode Command line tools will be installed.

$ brew install wxpython 
$ brew install rtl-sdr

# Plug in USB port a RTLSDR device
$ rtl_test -t
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
No E4000 tuner found, aborting.

$ sudo pip install rtlsdr-scanner
$ python -m rtlsdr_scanner

I hope it will help someone :-)
Rudolf

tggo, 2019-06-25:
not working with wxPython 4.0

> AttributeError: 'module' object has no attribute 'SystemSettings_GetMetric'

tggo, 2019-06-25:
also need  change 

> from wx.animate import AnimationCtrl, Animation

to

> from wx.adv import AnimationCtrl, Animation

cbr600, 2020-04-04:
the change to wx.adv fixed and issue I had on launch. I now have:

Traceback (most recent call last):
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/Library/Python/2.7/site-packages/rtlsdr_scanner/__main__.py", line 125, in <module>
    wx.Locale().Init2()
AttributeError: 'Locale' object has no attribute 'Init2'