EarToEarOak/RTLSDR-Scanner #56: freqs below 24mhz for some rtlsdr versions

luoGanda, 2019-03-18:
some rtlsdr have with(or without) upConverter ability to go down to 100khz or even below.
It seem that current software doesn't have an option to set "direct 'q' sampling mode".
So optional switch to select q mode and allowing freqs to be entered from 0 up would be perfect.
Or is there such setting that i am not aware of?

RtlSdr tools have this setting through direct_samp - which i think is 2 for direct q channel.

s53zo, 2019-06-25:
I have the same request. Can you please enable Direct Q sampling mode for RTL SDR V3 dongle?

As described on the webpage:
Experimental: HF Direct Sampling Mode - Listen to 500 kHz - 24 MHz with direct sampling. Simply connect an HF antenna to the SMA port, and choose the Q-branch mode.

luoGanda, 2019-07-02:
this was/is tested on windows7 version:

this devices have direct q channel selection in software(no harware mod etc, you just switch on another antenna port and enable direct-q branch), but, i searched on a webpage and manual, but there's nothing about it - neither the "Experimental: HF Direct ...".

The reason for the first post was primarily because i somehow could not select frequencies below some value - i think values below 50(or 30)mhz couldn't be set or something like this - but now they can be, i am not sure why - maybe i used some other device previously or something like this and this values went bogus.

However - i still don't know how to enable direct q branch mode - so that this frequencies can be used, in software(or on command line) i didn't find any switch for it.
For now, i'll just try to patch local py code.

Also, latest downloaded windowsVer 1.3.2 has version number 1.3.1 when opening about in software - or it's an mismatched windows download version.

luoGanda, 2019-07-02:
hurray..
yes, adding "self.sdr.set_direct_sampling(2)"(or "self.sdr.set_direct_sampling('q')) for a quick patch works,
now, only the settings needs to be updated somewhere..