EarToEarOak/RTLSDR-Scanner #65: Re opening DEVICE PERMISSIONS #43

Daddio61, 2020-07-15:
Here's my issue from #43.  On running in Windows 10 1803 Build 17134.1 I get the following response from Python 27\python.exe

Found Rafael Micro R820T tuner
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules

And when clicking on START in the GUI, Message box ERROR code -1 when opening SDR (device index =0)

I ran BBCR2.rfs after closing SDR# V1.0.0.1666, received the Python error, unplugged and reinserted the RTL SDR V3 then got the same message and then clicked START on the GUI to achieve the above repeatable error.

Great program thanks for coding it!!

You asked me to run the RTL-TEST.exe and I'm still having the same issues now as I did then.

You were going to run some more tests with Windows 10 and get back to me!

Mason10198, 2021-03-25:
I am also having this issue, except I am not getting any specific codes. It seems as though the application is trying to open the wrong device... all of my other SRD applications work fine, and I know for a face that my driver installation(s) are correct. The device window does not provide all of the correct information either.
.
[image]
.
.
The Error window is blank and provides no information.
.
[image]

danniell77, 2021-03-26:
Hello ! I have the same error "usb_open error -12, say if you solved the problem?
[image]
 Thanks!